sb-ocr/diy-spacemouse #1: spacemouse pro emulation USB driver

Arbel-arad, 2023-06-24:
someone already made software for getting a pi pico to report and function as a full spacemouse (with software support)
this should be easy to convert for your hardware and function without locking out the mouse.
[https://github.com/jfedor2/magellan-spacemouse](https://github.com/jfedor2/magellan-spacemouse)

MohrJonas, 2023-06-24:
Great idea on your part 😀!
Most of the firmware seems relatively simple, but I've opened an issue in the repo you linked, asking about some of the more delicate inner workings, which the creator gladly explained: https://github.com/jfedor2/magellan-spacemouse/issues/2

Arbel-arad, 2023-06-24:
i can do packet capture on my spacemouse (enterprise) if that will help.
last time i tried to make custom descriptors it didn't work out, but the original version of this seems to be written well.

MohrJonas, 2023-06-24:
Yeah my hope is that all that has to be done is to throw out the code responsible for reading from the magellan mouse via UART and replace it with code that reads from the magnetometer via I2C

Ryder17z, 2023-06-25:
There is also https://github.com/lukenuttall/SpaceNavigatorEmulator  which might be interesting as inspiration

Arbel-arad, 2023-06-25:
that does look interesting but it requires loading a custom DLL file.
that isn't great for security, and also won't work in my main CAD program (onshape) because it's web based.
they are also emulating an older version of the device that is no longer supported by new 3DXware driver versions.

Ryder17z, 2023-06-25:
Those area good points. But also begs the question of difficulty in getting this working with Solidworks

LoganFraser, 2023-07-07:
Any updates on this? I finally have the adjustments to the base design I wanted (2 buttons per side, easier 3d printing, qwiic wire routed though the center of the support post) so I will be assembling and testing this weekend.

Arbel-arad, 2023-07-07:
until i can get a pi pico there probably won't be much progress.
though i hear the raspberry pi availability is starting to recover now.

Ryder17z, 2023-07-07:
The pico is not as badly affected by the shortage though?

LoganFraser, 2023-07-07:
You don't need a pico, any rp2040 will work.  I'm using a Seeeduino Xiao RP2040 for mine, same footprint as the bom part just no qwiic socket.

MohrJonas, 2023-07-08:
I'm working on it occasionally. The USB parts seems to work alright, but I'm having trouble getting the pico to read the sensor

LoganFraser, 2023-08-19:
> I'm working on it occasionally. The USB parts seems to work alright, but I'm having trouble getting the pico to read the sensor

Did you make any progress on this?  This seems like a useful device somewhat hindered by mouse/keyboard emulation restrictions.

MohrJonas, 2023-08-19:
Unfortunately, no

Arbel-arad, 2023-08-20:
i received the rp2040 board, but i don't have the sensors to test with them.
i will check if the interface driver works anyway, but it won't be very useful.

BeardedCobra48, 2023-08-23:
Hei there :D 

Im also bit interested! :D 

I bought a old CadMan Space Mouse (I love the looks and feling of it! even more than the one of the new 3dconnexion i use at work!)
SHAME ON 3Dconnexion for killing old hardware just because they can! :(

I am wondering if I can get my CAD Man to work like a new 3d connexion with a emulator like this?

Also I would love this to get big and spread it all accross the internet!

there are a lot of old 3d space mouses around that are just not usable because 3d connexion decided to kill them!!

Lets tell everyone there is a solution and f*ck 3dconnexion in their a*ss !! damn they deserve it!!

Ryder17z, 2023-08-28:
@BeardedCobra48 
If I may be blunt: they do not even support modern products properly - why else would people be asking for over a decade to add dead-zone control? ([See thread from 2008](https://forum.3dconnexion.com/viewtopic.php?t=2120))

The current solution is really janky and involves modifying the config files for the official driver: https://forum.3dconnexion.com/viewtopic.php?t=16821   (but this applies to EVERY axis input)

It was not resolved in 2021 and I highly doubt it will ever be fixed officially.

I do not know if this project supports dead-zone control, but it would be awesome if it could be added with each axis independently adjustable, as per my own testing using multiple "Space Mouse Mini" on multiple computers has revealed it really needs little/none for up/down/rotation and some for the other axis; otherwise as soon as I take my hand off the device, it is almost guaranteed to fling the view a mile away from the object I'm working on in Solidworks.

RivasMario, 2023-09-12:
There is a value you can change in the code for dead zone. There is no official calibration test or setting for the spacemouse yet.

@LoganFraser how did you connect the seeduino without the qwik connector?

apadua, 2024-02-12:
> someone already made software for getting a pi pico to report and function as a full spacemouse (with software support) this should be easy to convert for your hardware and function without locking out the mouse. https://github.com/jfedor2/magellan-spacemouse

Can you tell me more about locking out the mouse? Are you saying when i deploy this code the normal mouse will stop moving? Have you tried working with it like this - does it break design flow significanly ( I assume it would)

A.