sb-ocr/diy-spacemouse #3: BOM

CyborgGearVR, 2023-06-25:
Please include a Bill of Materials when you add your Build Instructions.  That way we can purchase the same springs and boards you did,  Thanks.

jrsall92, 2023-06-26:
I made a BOM and a pull request, in the mean time until it's accepted, you can find it here https://github.com/sb-ocr/diy-spacemouse/pull/4

mpx1986, 2023-06-28:
Can you please add the size of the magnets

sb-ocr, 2023-06-28:
> I made a BOM and a pull request, in the mean time until it's accepted, you can find it here #4

Thanks! I was already working on this update. I uploaded the exact parts and hardware [BOM](https://github.com/sb-ocr/diy-spacemouse/tree/main/BOM)

closed 2023-06-28 by sb-ocr

CyborgGearVR, 2023-07-01:
Thank you for the quick action on this.  Starting to work on one now.

TheUltimateAbsol, 2023-07-01:
BOM not clear on the screws and heatsets, see https://github.com/sb-ocr/diy-spacemouse/issues/9