sb-ocr/diy-spacemouse #5: GY-521 sensor

axp-dev, 2023-06-30:
Hi, thanks for the project. You have inspired me to work. As in my country it is difficult to get the TLV493D, I will rework the inner triangle for the GY-521 sensor (MPU-6050). It will allow better position tracking.

When I come back from vacation, I will try to implement my ideas. And be sure to come here and show you the result.

V1s1S, 2023-07-08:
Привет, не поделишься переделанным проектом??

axp-dev, 2023-07-08:
@V1s1S на данный момент из готового это внутренний треугольник. Это единственное, что успел сделать и уехал до середины августа. Когда приеду напишу код для датчика.

[image]
[image]

Stout001, 2023-08-05:
Any progress on this. I also have plans using a MPU-6050/GY-521 and struggle with the code as I am also using a RP2040-Zero

dannybend, 2023-09-23:
Any progress on this?