sb-ocr/diy-spacemouse #6: Script python error

Emmanuele75, 2023-07-01:
Hi, great project but I get this error:

C:\Users\titan\Downloads\diy-spacemouse-main\diy-spacemouse-main\Code\Fusion360 Add-in\Send Home>SendHome.py
Traceback (most recent call last):
  File "C:\Users\titan\Downloads\diy-spacemouse-main\diy-spacemouse-main\Code\Fusion360 Add-in\Send Home\SendHome.py", line 1, in <module>
    import adsk.core, adsk.fusion, adsk.cam, traceback
ModuleNotFoundError: No module named 'adsk'

How can I solve this problem?
Thanks!

Kudzmat, 2023-07-01:
This error occurs when you're trying to access or use a module that cannot be found, which means it's probably not installed on your computer.

Try typing 'pip install adsk' in your terminal first. Also, make sure you don't misspell the name of the module when installing. I hope this helps.

Emmanuele75, 2023-07-02:
Thanks for the suggestion, this issue has been resolved.
Anyway I can't start the script, I get this error:

Failed:
Traceback (most recent call last):
  File "C:/Users/emman/AppData/Roaming/Autodesk/Autodesk Fusion 360/API/AddIns/SpaceMouse/SpaceMouse.py", line 18, in run
    buttonSample = cmdDefs.addButtonDefinition('HomeViewButton',
  File "C:\Users/emman/AppData/Local/Autodesk/webdeploy/production/dbacedcc6dabacdc41406088a765962c5f1923ad/Api/Python/packages\adsk\core.py", line 5939, in addButtonDefinition
    return _core.CommandDefinitions_addButtonDefinition(self, *args)
RuntimeError: 3 : the relative resourceFolder path not found

LoganFraser, 2023-07-08:
You have a C:\ then the rest are /, that's probably the issue.

blasterlabs, 2023-07-27:
> 

Hi, have you solve it? I've the same problem