sb-ocr/diy-spacemouse #8: A suggestion for wider compatibility

LoganFraser, 2023-07-01:
There are many versions of the pi tiny sized rp2040 board but most don't have the i2c socket on them.  For future revisions, could you keep the i2c pins clear (4 and 5) and put the buttons on different inputs?  That would make it easier to connect to more versions.  Of course, I can just edit the pin file, and I'll have to for my seeed xiao rp2040 anyway to move the i2c.  

Looking forward to the next version!  Thanks for sharing.  :)

sb-ocr, 2023-07-03:
Absolutely, there are some improvements that can be made for better printability, as you pointed out. I will definitely keep that in mind for the next iteration.

Regarding the buttons, I was thinking maybe the next one could have mini clusters of 2 or 3 on each side, leaving the "palm" side where you currently placed the third button clear, just because you rest your hand on it.

Thank you for the suggestions!

closed 2023-07-03 by sb-ocr

LoganFraser, 2023-07-04:
Tried out the 2 buttons per side thing.  I don't think there's room for 3, it starts getting really difficult into the rounded corners.  I had to move the mounting screws for the button plate 90 degrees.
[image]

I also added a wire path for the sensor down through the middle of the shaft.  
[image]
I should have the sensor board in a few days, it's the only piece I'm still missing.  Thanks again.