sb-ocr/diy-spacemouse #9: Size of the Heatset Insert?

TheUltimateAbsol, 2023-07-01:
Instructables says that it's a M2.5x3.4 heatset insert. I'm going to assume that 3.4 refers to the insert's outer diameter and not length.

When trying to order the part, it doesn't appear that 3.4 mm is a standard size, I can only find listings for 3.5 mm. Also I have no idea what length to order (3mm? 3.6mm? 4.0mm? 4.5mm? 5mm?)

The linked Amazon product only comes with M2.5x5, which doesn't even match the Instructables page.

Can I get a little bit of guidance here?

TheUltimateAbsol, 2023-07-01:
Furthermore the "M2.5 Socket Head Screws" are a bit confusing. The linked amazon item is for M3 Socket Head Screws and M2.5 Button Head Screws.

It appears Socket Head Screws are used in the video. Can I assume this link is incorrect?

sb-ocr, 2023-07-03:
You are right! It's a bit confusing. 

I use a localization service for the Amazon links, which is supposed to match exact products in different countries. When a product is out of stock or unavailable, it moves to the "equivalent." However, sometimes it completely misses the mark.

The inserts I used are indeed **M2.5 x 3.4mm** in length*. I had some left over from a previous build, and I got them from [McMaster Carr](https://www.mcmaster.com/94180A321/) I found the closest equivalent from amazon that I will update and fix the links manually. Lengthwise, you can also use 3mm or 3.5mm. Just make sure that the diameter is around 4.5mm, or modify the holes accordingly.

Same goes for the screws. The link for Amazon(US) was incorrect. I manually updated it to fix the issue. They are M2.5 Socket Head screws, as seen in the video.

Thanks for bringing this to my attention!

unlucio, 2025-08-31:
Here, I found what inserts to get, at a reasonable price:

[image]