sb-ocr/diy-spacemouse #12: 6 Axis Sugestion

david0green94, 2023-07-16:
Hi, 
How hard do you think it would be to modify the design for 6 axis? In the spirit of using parts laying around I realised a drone flight controller which are somewhat cheap and very available have at least 5 axis gyro + altitude. Just trying to work out how to pull the data from betaflight now.

LoganFraser, 2023-07-20:
That depends how big the flight controller pcb is, I'd imagine. Do you have a link for it?

sb-ocr, 2023-07-20:
I was playing with a similar idea by pairing the magnetometer readings with a gyro + accelerometer and calculating the Euler angles that way. That's why, in the first prototype, the sensor is placed on top of the platform. However, I found it a bit sticky and unreliable in terms of resolution for the amount of motion the knob is supposed to detect. It could be due to my poor programming, I only did a few quick tests, so I can't say for sure if t's the right approach or not.

stuart-warren, 2025-01-16:
There are a bunch of similar looking boards, I cant tell if they are actually useful here or not

* https://www.adafruit.com/product/4554
* https://www.adafruit.com/product/4754
* https://www.adafruit.com/product/4646