sb-ocr/diy-spacemouse #14: 3D Models for SLA Printing

maranhaoBruno, 2023-08-05:
Salim,

This is awesome, and I'd love to make my own, however; I have a SLA resin printer, and thus I cannot heat sink the threaded inserts, but have to press fit them in place. I was wondering if you could release STL files with the threaded insert positions increased to 4.5 mm in diameter. I tried to press fit some of the threaded inserts in and the holes are two small and the part ends up breaking apart. I opened the STEP file in FreeCAD, but I can't navigate down to the individual subelements of the larger parts, i.e. I can only get as far as "Spring Mech Top," "Sprint Mech Bottom," "Stem," and "Knob Shell" from the "Knob" part.

Thanks,
Bruno

RivasMario, 2023-08-15:
I believe he added the stl files. 

https://github.com/sb-ocr/diy-spacemouse/tree/main/CAD/STLs

sb-ocr, 2023-08-18:
You can download the model in a few different formats from this [link](https://a360.co/3CR37TL)

closed 2023-08-18 by sb-ocr