sb-ocr/diy-spacemouse #25: threads inserts issues

unlucio, 2025-08-30:
Hello everyone,

I've printed the parts, but I'm having an issue with the thread inserts:

1) Where can I find pictures or a diagram clearly pointing out all the 24 holes that need thread inserts?

2) I got 2.5M inserts, but all the holes are bigger than 3M 🥹

I printed it on an A1 mini.

unlucio, 2025-08-31:
I found the solutions to my problems, and I thought to share them for posterity.
Here are the inserts to buy, as well as pictures with all the holes that need them.

[image]
[image]
[image]

ilovecats4606, 2026-02-20:
> > Hello! I about to do this project, i needed to find these as well. The model you sent, in Amazon Australia its $75 (crazy expensive), do you know of any alternatives? thanks
> 
> You can look for something on Temu or Aliexpress.

Thanks, but i didnt know which one to look for specifically, I am quite new to making these sorts of stuff. Could you give a link? I would really appreciate it
Thanks